pachterlab/poseidon #10: Steppers are not independant !

Xqua, 2018-12-13:
Steps:

1) Set settings on the settings tab, example 
1: 0.1mm/s
2: 1mm/s
3: 10mm/s

2) Click send all settings

3) Go to syringe control

4) Click run 

The 3 steppers go at the same rotation speed !

Also, the numbers do not update on the Syringe control tab

sbooeshaghi, 2018-12-13:
Hey Xqua,

A couple of things, if you have your microstepping jumpers placed on your CNC sheild board, you need to specify the microstepping amount (1/32, 1/16 etc) in the python code under the unit conversion functions (we have not automated this yet). Also the range of speeds for the carriage is limited by the microstepping that you select. For example if you select 1/32 microstepping you wont be able to go 5mm/s. We don't have exact speed ranges yet for the different microsteps but we are working on it. Also make sure also that you are using stepper motors with 200 steps as that will affect the speed.

Second there is a known bug that we are working on fixing and that is the following: the very first time you press "Run" or "Jog +/Jog-" after you send your settings, then the speed of the motors is not accurate. A temporary workaround is to send your settings, then press "Jog +", then immediately press stop. Now if you press "Jog" or "Run" the steppers should run at the appropriate speed. Keep in mind you will have to do this after every speed change. (yes this is annoying and yes we will fix it :) )

Lastly, we have set the acceleration for the motors to a fixed value in the arduino code in this release as we were having some issues with varying the acceleration from the GUI.

I hope that helps! Please let me know if there are any other problems that arise.

Sina B.

Xqua, 2018-12-13:
Hi ! 

Yes this already helps (I think ?)
Alright, so I followed the poseidon tutorial, a bought the steppers you recommended (so there should not be any discrepancy here)

So for example I set this
[image]

Then went here, clicked Jog + then Stop
Then set as below

[image]

What I am expecting is for pump 3 to run 1 second, pump 2 for 2 sec, and pump 1 for 10 sec right ? 
(1 mm each with different speed)

But the 3 will run for about 30 seconds 

And at about the same speed (visually ... )

Xqua, 2018-12-13:
(Also I have experience with python and arduino so I'll be happy to help with those bugs ^^ but I was just trying an "out of the box" trial)
Also were in the code is this unit conversion 1/32 ?

this line ? `mm = steps/200/32*0.8` 
Because it's already set ! :D

Xqua, 2018-12-13:
also I believe this is probably a typo ?
[image]

Where this should be 

      stepper1.move(p1_distance_left);
      stepper2.move(p2_distance_left);
      stepper3.move(p3_distance_left);
      while (stepper1.distanceToGo() >= 0 && stepper2.distanceToGo() >= 0 && stepper3.distanceToGo() >= 0) {

sbooeshaghi, 2018-12-14:
Thanks for the info that helps out! Can you do the following for me, can you comment out `line 1104` and uncomment `line 1105`, and comment out `line 1118` and uncomment 1119 in poseidon_main.py using  the hashtag "#"?

You then have two options to run the software, you can recompile the executable or you can go to your terminal and run `python3 poseidon_main.py`

Then try running your settings again and let me know if that works!

Sina B.

sbooeshaghi, 2018-12-14:
> also I believe this is probably a typo ?
> [image]
> 
> Where this should be
> 
> ```
>   stepper1.move(p1_distance_left);
>   stepper2.move(p2_distance_left);
>   stepper3.move(p3_distance_left);
>   while (stepper1.distanceToGo() >= 0 && stepper2.distanceToGo() >= 0 && stepper3.distanceToGo() >= 0) {
> ```

Yes there is! I'll fix it. Thank you :)

Xqua, 2018-12-14:
Ahahah yes ! this 32 divisions was the trick !

Xqua, 2018-12-14:
All good now, you should prob change this in the github code ;)

closed 2018-12-14 by Xqua

sbooeshaghi, 2018-12-14:
Great! Glad it worked out. I appreciate you trying it "out of the box" it helps identify these bugs :)

I'll go ahead and make those changes now.

Sina B.