pachterlab/poseidon #12: GUI number not updating

Xqua, 2018-12-14:
Alright so moving on to the next bug I guess. 

[image]

This part of the GUI is not updating ? Whatever one does, it either stays to 0 or display some cryptic (.59??) number for pump 1 only

sbooeshaghi, 2018-12-14:
This feature has not been successfully implemented yet, maybe I should remove it from the GUI so as to not confuse users.

I was having an issue with the amount of Serial Prints I was sending from the Arduino to display these positions so I left it at that..

Sina B.

Xqua, 2018-12-14:
Hum ... 
Could you not do an estimate ? 
As the arduino is going to run until a new command is sent, it should take milliseconds for the command to go through. 
Therfore you can either measure it, or just assume t = 0 and then run an estimate based on the speed of the motor ? 

And send from the arduino the current position every ... 10 / 100 loop ? just to adjust ?

sbooeshaghi, 2018-12-17:
Hmm not a bad idea! I'll play around with it and see what sort of results I get. Thank you again for the suggestions!

Sina B.