pachterlab/poseidon #29: Size of the pump for 3D printing & standard 3D printer bed size

jcmolloy, 2019-10-03:
I am printing off the component parts but I only have access to a printers with a build base of 22 cm (Ultimaker, Prusa i3, Ender 3). The STL is 25 cm - any suggestions other than finding someone with a bigger printer? 

I looked at the model vertically as it will fit into the Ultimaker Extended but seems like a brutal overhang! It would be great if you could confirm your printing strategy before I try and end up with a mess. Thanks very much!

sbooeshaghi, 2019-10-03:
We have not tried printing the pump vertical with that overhand, but I don't see why it shouldn't work! All of our pumps have been printed on a Makergear M2.

You may know this but there are services that offer to print parts for quite cheap. This may be something to look in to.

Sina B.

closed 2019-10-03 by sbooeshaghi

jcmolloy, 2019-10-04:
Thanks @sbooeshaghi. I figured as I have access to eight types of 3D printer and a lot of filament I should be able to make it work without outsourcing 🙂. Ultimaker recommend a max 1-2mm unsupported 90 degree overhangs so if you haven't successfully tried then I don't think it's likely to work. I'll redesign the Fusion360 model to fit a 22cm bed and I'll report back and share the file if it all works out.