pachterlab/poseidon #33: The commands were not sent to stepper motors; however, the board connected to the gui controller successfully

maydanshahi, 2020-10-06:
I have followed all structures but The commands were not sent to stepper motors; however, the board connected to the gui controller successfully. I validated the arduino uno, cnc shield and stepper motors with another firmware (grbl) and every thing was okay. The commands sent to the board and then stepper motors successfully but when I use the poseidon structures, I faced to this problem. The controller was connected to board successfully, and the commands were sent to the arduino board correctly, but the stepper motors were not work. What should I do? could anyone help me?

gallagherk17, 2021-04-12:
Hey did you ever figure this out? I am facing a similar issue.

maydanshahi, 2021-04-27:
Hi
Not yet

Mubeen5611, 2023-03-18:
Is you resolve that issue then plz guide me

Mubeen5611, 2023-03-18:
Command was not sent to motor plz guide