pachterlab/poseidon #37: Commands are not reflected in motor movement

gallagherk17, 2021-04-13:
**Describe the bug**
A clear and concise description of what the bug is.
I have connected to the controller and have sent the settings but when I try to jog or run the motor nothing happens.  The code seems to be sending to the Arduino but nothing is occurring as the output

**To Reproduce**
Steps to reproduce the behavior:
1) Set settings to:
[image]
2) Try to Jog + (10mm)
3) No response from motor

No error is populated in code
[image]

**Expected behavior**
A clear and concise description of what you expected to happen.
I would expect the motor would jog forward about 10mm 

**Screenshots**
If applicable, add screenshots to help explain your problem.
My Arduino set up
[image]

**Software versions (please complete the following information):**
 - OS: Windows
- Python version 3.6

**Additional context**
I am new to arduinos but I have been trying to follow all the steps you have laid out.  Any help on how to solve this issue would be greatly appreciated.

sbooeshaghi, 2021-04-15:
Have you placed jumpers underneath the motor controller (in the CNC shield) to correspond to 32 microstepping?

Mubeen5611, 2023-03-18:
> Have you placed jumpers underneath the motor controller (in the CNC shield) to correspond to 32 microstepping?

Is you resolve that issue plz guide me

Mubeen5611, 2023-03-18:
> **Describe the bug** A clear and concise description of what the bug is. I have connected to the controller and have sent the settings but when I try to jog or run the motor nothing happens. The [...]

Is you resolve that problem